OllisGit/OctoPrint-PrintJobHistory #168: Is it possible to have the Image and Thumbnail?

leoheck, 2021-08-21:
Is it possible to allow both? I would use the thumbnail if it exists since it is better to spot differences in the part that was print. But I would also like to have the picture too, to check how the print job was.

My suggestion is to have it as the tool tip when the user pass the mouse over the image.
So, if it selects to show thumbnails, the tool tip of the thumbnail would be the picture. 

Fit it chooses to use the picture as the default, the tool tip would be the thumbnail.

Having a picture would be good in case of failure too, so it would be a recording of the state/height of the print when that the fail has occurred.

OllisGit, 2021-11-13:
Hi @leoheck,

fyi: it is already on my mid-term roadmap

leoheck, 2022-01-23:
Oh, come on github-actions!

OllisGit, 2022-01-24:
fyi: I am already started with the feature "multi images/cameras". for that reason I setup my RasPI with the PI-Cam and also with a USB-Cam (see https://github.com/OllisGit/OctoPrint-KnowledgeBase/wiki/OctoPiTipps#setup-raspicam-and-usb-cam-endoscop).

To support multi-cams I wrote a "Pull-Request" for Multi-Cam-Support: https://github.com/mikedmor/OctoPrint_MultiCam/pull/43
But I think the Plugin is not maintained...or he did a 1year octoprint-sabbatical. 

And I started evaluating UI-Widgets for displaying multiple images:
 
[image]