OllisGit/OctoPrint-PrintJobHistory #170: Feature request : Auto-add to ics file after print

pracovnik, 2021-09-22:
I would like to ask, if there could be an auto-export-append into ics file after print.

Backstory:
I have in octoprint end gcode a (plugin: action commands) python script, that check logs and with data gathered, it export it into a icalendar file (ics) one entry is append.  I have shown it to few people on FB and they told me that they want it too. As I think, this is the most near plugin that has all the information, I wanted to ask it as a feature request.

Via badly configured haproxy, I have the ics file exposed to local network. And with ical client, on PC MS Outlook, on Android ics sync, shown as calendar.

[image]

[log_to_ical.py.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7210935/log_to_ical.py.txt)

OllisGit, 2021-11-13:
Hi @pracovnik,

if I understand you correct you want an other export functionality. Beside the CSV-Export, there should be ics-File export.
The export file include all selected printjobs as an event entry, right?
[image]
 
In the next version there will be a print-job-template where the user can design a printing-layout in a jinja2 file and upload it to OP.
I think it will be easy to implement the same feature for the ics-feature.
The ics-template could look like this (pseudo-code):
```
BEGIN:VCALENDAR
VERSION:2.0
PRODID:-//ical.marudot.com//iCal Event Maker
CALSCALE:GREGORIAN
BEGIN:VTIMEZONE
....
foreach(job in allPrintJobs
  BEGIN:VEVENT
  DTSTAMP:20211113T110702Z
  UID:1636801614876-36929@ical.marudot.com
  DTSTART;TZID=Europe/Berlin:20211122T120000
  DTEND;TZID=Europe/Berlin:20211122T120000
  SUMMARY:Print of {{job.printJobName}}
  DESCRIPTION:Print used: ....
  END:VEVENT

...
END:VCALENDAR
```

pracovnik, 2021-11-16:
@OllisGit 
yes, exactly. 

Will it also have something like automatic export based on template/ics after each print ?

closed 2022-02-27 by github-actions[bot]