OllisGit/OctoPrint-PrintJobHistory #173: Spool not saved with Arc Welder

vogler, 2021-11-03:
I don't know which plugin needs fixing, but whenever I enable Arc Welder, all the spool and used material fields are empty.

PrintJobHistory (1.14.0)
SpoolManager Plugin (1.5.0)
Arc-Welder (1.0.0+u.bb71e8f)

Arc Welder Setting `Overwrite Source File` is enabled.
Maybe it's a data race (starts analyzing, Arc Welder overwrites file)?

OllisGit, 2021-11-13:
Hi @vogler,
please attach the octoprint.log maybe there is something that could identify the issue.

vogler, 2021-11-13:
The original log is gone. I checked it at the time but didn't see anything obvious.
I just tested it again. Enabled Arc Welder, overwrite enabled.
First I printed some existing .gcode file - the material fields were there.
Then I sent a new print from Cura, Arc Welder processed it, and the print started, but the info box on the left already said 'Required Filament unknown'. Cancelled after some time - material fields empty.

`plugin_arc_welder.log` is empty.
Here's the whole `octoprint.log` of the runs described above:
<details>
  <summary>Click to expand!</summary>
  
  ```
2021-11-13 11:49:25,087 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-11-13 11:49:25,089 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2021-11-13 11:49:25,089 - octoprint.server - INFO - OctoPrint 1.7.2
2021-11-13 11:49:25,091 - octoprint.plugin.core - INFO - 57 plugin(s) registered with the system:
| !Access Anywhere - The Spaghetti Detective (1.7.2) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_thespaghettidetective
|  Action Command Notification Support (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Active Filters Extended (0.1.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_active_filters_extended
|  Announcement Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Arc-Welder (1.0.0+u.bb71e8f) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_arc_welder
|  Autoscroll (0.0.3) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_autoscroll
|  Autoselect Plugin (0.3.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_autoselect
|  Backup & Restore (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.1.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  Calibration-Companion (1.0.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_calibrationcompanion
|  Cancel Objects (0.4.6) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_cancelobject
|  Consolidated Tabs (1.0.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_consolidatedtabs
| !Continuous Print (1.1.2) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/continuousprint
|  Core Wizard (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Cost Estimation (3.3.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_costestimation
|  Cura Thumbnails (1.0.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_ultimakerformatpackage
| !Dashboard (1.14.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.27.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Dragon Order (0.1.7) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_dragon_order
|  Error Tracking (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/eventmanager
| !Filament Manager (1.8.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_filamentmanager
|  File Check (2021.2.23) (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_file_check
| !FileManager (0.1.6) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_filemanager
|  Firmware Check (2021.10.11) (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_firmware_check
|  Firmware Updater (1.12.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_firmwareupdater
|  GCode Viewer (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  GcodeEditor (0.2.12) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_GcodeEditor
|  Logging (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/logging
|  Marlin GCode Documentation (0.14.0) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_marlingcodedocumentation
|  MeatPack (1.5.21) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/OctoPrint_MeatPack
|  MQTT (0.8.10) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_mqtt
|  MultiCam (0.2.8) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_multicam
| !OctoEverywhere! (0.2.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_octoeverywhere
|  Octolapse (0.4.1) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.10.28) (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
[... cut ...]
2021-11-13 21:19:09,599 - octoprint.plugins.DisplayLayerProgress - INFO - Read total height from MetaFile
2021-11-13 21:19:12,329 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2021-11-13 21:19:25,113 - octoprint.server.heartbeat - INFO - Server heartbeat <3
  ```
</details>

For the second print it says `2021-11-13 21:19:05,669 - octoprint.plugins.PrintJobHistory - INFO - There is no calculated filament data in meta-file?`.

(Aside: the `Database connection` logging seems excessive: 162 matches for `password` just in this log.)

OllisGit, 2021-11-13:
Hi @vogler, 
the plugin reads some job-statistics from the metadata-file (```.metadata.json```) located in the upload-folder.
Please take a look if you can find this section for you print-file (in particular "filament"):
```
        "analysis": {
            "printingArea": {
                "maxX": 127.881,
                "maxY": 95.506,
                "maxZ": 10.9,
                "minX": 88.081,
                "minY": 42.554,
                "minZ": 0.3
            },
            "dimensions": {
                "depth": 52.952,
                "height": 10.6,
                "width": 39.8
            },
            "estimatedPrintTime": 2754.571579330689,
            "filament": {
                "tool0": {
                    "length": 2016.75497,
                    "volume": 0.0
                }
            }
        }
```
If it is not available, it could be deactivate, please take a look into the feature-plugin-settings:
[image]

Btw I need to write down where all the data came from, because depending on the installed plugins you will see different values.
Btw2. the ```Database connection``` is annoying and I will remove this in one of the next releases.

vogler, 2021-11-13:
I have no `.metadata.json` files.
Changed the setting from `Only when idle (not printing)` to `Always`.
Still no such files, but now `Required Filament unknown` is gone and it shows the estimates.
Cancelled the print and the material fields are no longer empty 👍 

Any conclusions how to improve this? Seems broken.
Should Arc Welder delay printing until it is done pre-processing?
Should this plugin wait for the metadata even if this setting is set to `Only when idle (not printing)`?
I assume the intent behind the setting is to not analyze a lot of other gcode, but shouldn't it always analyze the current print?

OllisGit, 2022-01-16:
Hi @vogler,
I never used arcwelder - plugin, but I think there is also a Cura-Plugin for it. So maybe you should use arcewelder in cura first.

Because the missing metadata is also related to other support-issues, I will try to improve the "communication with the user" about the missing metadata...or maybe wait or retrigger the job.

vogler, 2022-01-18:
Yes, that's what I ended up using instead. IMO anyway a better fit to have it in the slicer with all the other settings.