OllisGit/OctoPrint-PrintJobHistory #174: All history gone after updating, error while saving the most recent print

jakezilla, 2021-11-13:
PJH said it could not save history, and to check my .log
Also, all my history is gone now

Here is the PJS section from the .log:

```
2021-11-13 16:51:02,669 - octoprint.plugins.PrintJobHistory.DatabaseManager - ERROR - Could not insert printJob into database:no such table: pjh_costmodel
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3144, in execute_sql
    cursor.execute(sql, params or ())
sqlite3.OperationalError: no such table: pjh_costmodel

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/DatabaseManager.py", line 511, in insertPrintJob
    if (printJobModel.getCosts() != None):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/models/PrintJobModel.py", line 46, in getCosts
    if (self.costs != None and len(self.costs) > 0):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2008, in __len__
    self._ensure_execution()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1990, in _ensure_execution
    self.execute()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1907, in inner
    return method(self, database, *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1978, in execute
    return self._execute(database)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2150, in _execute
    cursor = database.execute(self)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3157, in execute
    return self.execute_sql(sql, params, commit=commit)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3151, in execute_sql
    self.commit()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2917, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 190, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3144, in execute_sql
    cursor.execute(sql, params or ())
peewee.OperationalError: no such table: pjh_costmodel
2021-11-13 16:51:02,671 - octoprint.plugins.PrintJobHistory - ERROR - PrintJob not captured, see previous error log!
2021-11-13 16:51:02,671 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2021-11-13 16:51:02,703 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/loadPrintJobHistoryByQuery [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3144, in execute_sql
    cursor.execute(sql, params or ())
sqlite3.OperationalError: no such table: pjh_costmodel

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 284, in get_printjobhistoryByQuery
    allJobsAsDict = TransformPrintJob2JSON.transformAllPrintJobModels(allJobsModels, self._file_manager)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 85, in transformAllPrintJobModels
    jobAsDict = transformPrintJobModel(job, fileManager, deleteDateTimeFromDict)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 58, in transformPrintJobModel
    costs = job.getCosts()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/models/PrintJobModel.py", line 46, in getCosts
    if (self.costs != None and len(self.costs) > 0):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2008, in __len__
    self._ensure_execution()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1990, in _ensure_execution
    self.execute()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1907, in inner
    return method(self, database, *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1978, in execute
    return self._execute(database)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2150, in _execute
    cursor = database.execute(self)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3157, in execute
    return self.execute_sql(sql, params, commit=commit)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3151, in execute_sql
    self.commit()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2917, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 190, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3144, in execute_sql
    cursor.execute(sql, params or ())
peewee.OperationalError: no such table: pjh_costmodel
2021-11-13 16:51:05,154 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2021-11-13 16:51:56,014 - octoprint.server.heartbeat - INFO - Server heartbeat <3
```

jakezilla, 2021-11-13:
OctoPrint 1.7.2 Python 3.7.3 OctoPi 0.17.0 PJH 1.15.0

shamlian, 2021-11-14:
Same problem on my installation (OctoPrint 1.7.2, Python 3.7.3, PJH 1.15.0). Log follows:

```
2021-11-13 23:23:17,988 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2021-11-13 23:23:17,989 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job...
2021-11-13 23:23:17,991 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-11-13 23:23:18,128 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-11-13 23:23:18,130 - octoprint.plugins.PrintJobHistory - INFO - Try reading Thumbnail
2021-11-13 23:23:18,134 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Thumbnail is present
2021-11-13 23:23:18,134 - octoprint.plugins.PrintJobHistory - INFO - Thumbnail was captured from metadata
2021-11-13 23:23:18,134 - octoprint.plugins.PrintJobHistory - INFO - Try reading Thumbnail
2021-11-13 23:23:18,136 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try converting thumbnail '/home/pi/.octoprint/data/PrintJobHistory/../UltimakerFormatPackage/CE3_rounded_cube_module.png' to '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20211113-230920.jpg'
2021-11-13 23:23:18,158 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Converting successfull!
2021-11-13 23:23:18,159 - octoprint.plugins.PrintJobHistory - INFO - Thumbnail was captured from metadata
2021-11-13 23:23:18,159 - octoprint.plugins.PrintJobHistory - INFO - Try reading filament
2021-11-13 23:23:18,176 - octoprint.plugins.PrintJobHistory - INFO - Costs could not captured, because CostEstimation-Plugin not installed/enabled
2021-11-13 23:23:18,176 - octoprint.plugins.PrintJobHistory - INFO - Try storing printjob model
2021-11-13 23:23:18,196 - octoprint.plugins.PrintJobHistory.DatabaseManager - ERROR - Could not insert printJob into database:no such table: pjh_costmodel
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3144, in execute_sql
    cursor.execute(sql, params or ())
sqlite3.OperationalError: no such table: pjh_costmodel

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/DatabaseManager.py", line 511, in insertPrintJob
    if (printJobModel.getCosts() != None):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/models/PrintJobModel.py", line 46, in getCosts
    if (self.costs != None and len(self.costs) > 0):
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2008, in __len__
    self._ensure_execution()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1990, in _ensure_execution
    self.execute()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1907, in inner
    return method(self, database, *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 1978, in execute
    return self._execute(database)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2150, in _execute
    cursor = database.execute(self)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3157, in execute
    return self.execute_sql(sql, params, commit=commit)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3151, in execute_sql
    self.commit()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 2917, in __exit__
    reraise(new_type, new_type(exc_value, *exc_args), traceback)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 190, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 3144, in execute_sql
    cursor.execute(sql, params or ())
peewee.OperationalError: no such table: pjh_costmodel
2021-11-13 23:23:18,215 - octoprint.plugins.PrintJobHistory - ERROR - PrintJob not captured, see previous error log!
```

spanzetta, 2021-11-14:
same problem here.. all history gone (only statistic reports all data)

LUFEZ, 2021-11-14:
I think the reason is the database scheme upgrade.

```
2021-11-14 08:08:22,780 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - We need to upgrade the database scheme from: '6' to: '7'
2021-11-14 08:08:22,800 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Database migration from '6' to '7'
2021-11-14 08:08:22,800 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO -  Starting 6 -> 7
2021-11-14 08:08:22,803 - octoprint.plugins.PrintJobHistory.DatabaseManager - ERROR - Error during database upgrade!!!!
2021-11-14 08:08:22,803 - octoprint.plugins.PrintJobHistory.DatabaseManager - ERROR - near "DROP": syntax error
```

```
def _upgradeFrom6To7(self):
...
ALTER TABLE "pjh_filamentmodel" DROP COLUMN "spoolCostUnit";
...
```

"DROP COLUMN" is not supported by SQLite 3.27.2 installed on my RaspberryPI

OllisGit, 2021-11-14:
First: Don't worry "History Data" is not gone! 
If the database needs a patch (like in this release) a backup is created.
Something like this:
```data/PrintJobHistory/printJobHistory-backup-20211112-1921-V6.db```

You have two options:
- Wait for new release that fix the issue (preferred)
- Connect to ```data/PrintJobHistory/printJobHistory.db``` via a database-client and execute the following sql script by yourself and restart the server
```
CREATE TABLE "pjh_costmodel" ("databaseId" INTEGER NOT NULL PRIMARY KEY,
"created" DATETIME NOT NULL,
"printJob_id" INTEGER NOT NULL,
"totalCosts" REAL,
"filamentCost" REAL,
"electricityCost" REAL,
"printerCost" REAL,
"otherCostLabel" VARCHAR(255),
"otherCost" REAL,
"withDefaultSpoolValues" INTEGER,
FOREIGN KEY ("printJob_id") REFERENCES "pjh_printjobmodel" ("databaseId") ON DELETE CASCADE);

UPDATE 'pjh_pluginmetadatamodel' SET value=7 WHERE key='databaseSchemeVersion';
```
The ALTER command is not needed at runtime, it is just a "cleanup command"

OllisGit, 2021-11-14:
Sidenote: 

I looked into the SQLite Documentation and there is no hint about the minimum version needed for this command. Only for ALTER TABLE RENAME. (https://sqlite.org/lang_altertable.html#altertabdropcol)

Also, in the change log the drop command was mentioned in release 3.35.0...but I am using version 3.30.1 on my system...strange
(https://sqlite.org/changes.html)

OllisGit, 2021-11-14:
New version is released (1.15.1), please try and give me a feedback.

Thanks in advance
Olli

spanzetta, 2021-11-14:
Hi Olli, I was sure that nothing was lost.. as I reported the statistics was there so..
You was very very fast to find the issue and solve it.. It was fantastic if all programmers was like you ;-)
I will test 1.15.1 as soon as my current print will end..
Thanks a lot for your great job!

spanzetta, 2021-11-14:
I confirm that 1.15.1 shows again all jobs.. I've just lost the last print job before the update..
Good job!
Thanks again

donlook, 2021-11-14:
I had a new print that does not show in the history list. Here is the trace information from the octoprint.log:
(I redacted the user and password from the database connection information.)

`2021-11-14 12:08:19,774 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 12:08:19,774 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'XXXX', 'password': 'XXXX', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 12:08:19,779 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 12:08:19,799 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 976, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 753, in _printJobFinished
    self._addCostsToPrintModel(self._currentPrintJobModel)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 684, in _addCostsToPrintModel
    costData = self._costEstimationPluginImplementation.api_getCurrentCostsValues()
AttributeError: 'CostEstimationPlugin' object has no attribute 'api_getCurrentCostsValues'
`

OllisGit, 2021-11-14:
Hi @donlook,

you need at least CostEstimation 3.4.0+.

closed 2021-11-14 by OllisGit

vogler, 2021-11-17:
@donlook see https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/175#issuecomment-970980137

I also see no more history after I manually added the lost print job.
However, here it's the result of a type error somewhere:
~~~
Traceback (most recent call last):
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 284, in get_printjobhistoryByQuery
    allJobsAsDict = TransformPrintJob2JSON.transformAllPrintJobModels(allJobsModels, self._file_manager)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 85, in                               transformAllPrintJobModels
    jobAsDict = transformPrintJobModel(job, fileManager, deleteDateTimeFromDict)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 22, in transformPrintJobModel
    fileSizeFormatted = StringUtils.get_formatted_size(fileSize)
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/common/StringUtils.py", line 229, in get_formatted_size
    if num < 1024.0:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
~~~

vogler, 2021-11-17:
Fixed for now  by setting `fileSize` manually:
```
sqlite> update pjh_printjobmodel SET fileSize = 0 where databaseId = 195;
```
Without fixing the code, manually adding a print job will break showing them.

vogler, 2021-11-21:
@OllisGit can you reproduce `fileSize` missing being a problem when manually adding a job?