OllisGit/OctoPrint-PrintJobHistory #175: Improve Plugin-Dependency Version detection

OllisGit, 2021-11-14:
Currently there is only a "available-check" instead of available plus version number.
E.g. costs-feature is only working with CostEstimation 3.4.0+

neolao, 2021-11-15:
but CostEstimation is only at 3.3.0 ... https://github.com/OllisGit/OctoPrint-CostEstimation

vogler, 2021-11-17:
Just lost the data for a print job because of this:

~~~
2021-11-17 01:11:24,325 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)                                                                                         
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)                                                                                         
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/__init__.py", line 976, in on_event
    self._printJobFinished("success", payload)                                                                                                       
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/__init__.py", line 753, in _printJobFinished
    self._addCostsToPrintModel(self._currentPrintJobModel)                                                                          
  File "/home/pi/OctoPrint/lib/python3.9/site-packages/octoprint_PrintJobHistory/__init__.py", line 684, in _addCostsToPrintModel
    costData = self._costEstimationPluginImplementation.api_getCurrentCostsValues()
AttributeError: 'CostEstimationPlugin' object has no attribute 'api_getCurrentCostsValues'
~~~

Since there's no release tagged:
`bin/pip install "https://github.com/OllisGit/OctoPrint-CostEstimation/archive/refs/heads/master.zip"`

OllisGit, 2021-11-19:
tracked here #179 ..has nothing (or slightly) to do version-check

OllisGit, 2021-11-19:
The latest release 1.15.2 checks the 3rd party plugins with the needed version, instead of just the availability.

closed 2021-11-19 by OllisGit