OllisGit/OctoPrint-PrintJobHistory #178: Print Job Overview fails to display anything if a job has no file size

hvraven, 2021-11-17:
For testing I added a job with minimal entries, that apparently also meant no file size. Now the overview fails to display anything as long as the job is within the selected query range.

Error section from the `octoprint.log`
```
2021-11-17 22:16:31,977 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/loadPrintJobHistoryByQuery [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 284, in get_printjobhistoryByQuery
    allJobsAsDict = TransformPrintJob2JSON.transformAllPrintJobModels(allJobsModels, self._file_manager)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 85, in transformAllPrintJobModels
    jobAsDict = transformPrintJobModel(job, fileManager, deleteDateTimeFromDict)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/TransformPrintJob2JSON.py", line 22, in transformPrintJobModel
    fileSizeFormatted = StringUtils.get_formatted_size(fileSize)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/common/StringUtils.py", line 229, in get_formatted_size
    if num < 1024.0:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
```

OllisGit, 2021-11-19:
Hi @hvraven,

bug is fixed in latest release 1.15.2
Please test and give feedback.

Thx,
Olli

hvraven, 2021-11-20:
Hi Olli,

thanks for the quick fix, it does solve the problem.

Cheers,
Hendrik

closed 2021-11-20 by hvraven