OllisGit/OctoPrint-PrintJobHistory #179: Dependency problem? CostEstimation (Wrong Version, expected: 3.4.0+ current: 3.3.0)

vonProteus, 2021-11-18:
hi
when i login to octopi i get 
```
To get the real power of the Print Job History - Plugin you need the following plugins installed/enabled:

CostEstimation (Wrong Version, expected: 3.4.0+ current: 3.3.0)
```
this is normal?

i have installed 
<details>
<summary><b>plugin_list.json</b></summary>

```
[
  {
    "key": "touchui",
    "name": "TouchUI",
    "url": "https://github.com/BillyBlaze/OctoPrint-TouchUI"
  },
  {
    "key": "prusaslicerthumbnails",
    "name": "Slicer Thumbnails",
    "url": "https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails"
  },
  {
    "key": "PrusaETAOverride",
    "name": "Prusa ETA override Plugin",
    "url": "https://github.com/kanocz/octopi_eta_override"
  },
  {
    "key": "PrintJobHistory",
    "name": "PrintJobHistory",
    "url": "https://github.com/OllisGit/OctoPrint-PrintJobHistory"
  },
  {
    "key": "preheat",
    "name": "Preheat Button",
    "url": "https://github.com/marian42/octoprint-preheat"
  },
  {
    "key": "UltimakerFormatPackage",
    "name": "Cura Thumbnails",
    "url": "https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage"
  },
  {
    "key": "telegram",
    "name": "Telegram Notifications",
    "url": "https://github.com/fabianonline/OctoPrint-Telegram"
  },
  {
    "key": "SpoolManager",
    "name": "SpoolManager Plugin",
    "url": "https://github.com/OllisGit/OctoPrint-SpoolManager"
  },
  {
    "key": "prometheus_exporter",
    "name": "Prometheus Exporter Plugin",
    "url": "https://github.com/tg44/OctoPrint-Prometheus-Exporter"
  },
  {
    "key": "prettygcode",
    "name": "PrettyGCode",
    "url": "https://github.com/Kragrathea/OctoPrint-PrettyGCode"
  },
  {
    "key": "octopod",
    "name": "OctoPod Plugin",
    "url": "https://github.com/gdombiak/OctoPrint-OctoPod"
  },
  {
    "key": "ipOnConnect",
    "name": "ipOnConnect",
    "url": "https://github.com/jneilliii/OctoPrint-ipOnConnect"
  },
  {
    "key": "firmwareupdater",
    "name": "Firmware Updater",
    "url": "https://github.com/OctoPrint/OctoPrint-FirmwareUpdater"
  },
  {
    "key": "filemanager",
    "name": "FileManager",
    "url": "https://github.com/Salandora/OctoPrint-FileManager"
  },
  {
    "key": "excluderegion",
    "name": "Exclude Region",
    "url": "https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin"
  },
  {
    "key": "costestimation",
    "name": "Cost Estimation",
    "url": "https://github.com/OllisGit/OctoPrint-CostEstimation"
  },
  {
    "key": "BetterHeaterTimeout",
    "name": "BetterHeaterTimeout",
    "url": "https://github.com/tjjfvi/OctoPrint-BetterHeaterTimeout"
  },
  {
    "key": "octolapse",
    "name": "Octolapse",
    "url": "https://github.com/FormerLurker/Octolapse"
  },
  {
    "key": "DisplayLayerProgress",
    "name": "DisplayLayerProgress Plugin",
    "url": "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress"
  }
]
```

</details>
and
 
```
OctoPrint version : 1.7.2
OctoPi version    : 0.18.0
```

as far as i can say i have everything uptodate

OllisGit, 2021-11-19:
Hi @vonProteus,
 
uppps....there was something wrong with the release pipeline of the cost-plugin (now fixed and Version 3.4.0 is out now).

Surprisingly NO ONE reported that fact ;-)

If you install the lastest version via the Plugin-Manager you can ignore the Version-Tag in the bottom. It takes some time to "sync" the latest number.
After installation the version 3.4.0 is on your system and can be used together with PrinJobHistory.

[image]

OllisGit, 2021-11-19:
Hi @vonProteus,

in the latest release 1.15.2 I changed the 3rd party plugin check. The version-numer is now also compared.
Please test and give feedback.

Thx,
Olli

vonProteus, 2021-11-21:
yes everything is fine after updating to 1.15.2 and 3.4.0

closed 2021-11-21 by vonProteus