OllisGit/OctoPrint-PrintJobHistory #180: PJH collects wrong information fron Filament Manager

schumsel293, 2021-11-22:
PJH always collects informations fron Filament Manager (or PRUSA MMU Filament Selector?).
when printing is done, PJH always takes the values from Tool 0, although Tool 1 had been used for the print.

Similar results wit Spool Manager (not installed parallel to Filament Manager)
Nevertheless PJH decreases the material usage from the used filament (Tool 1) in a correct way.

I would be happy with a response from you

I added 3 screenshots documenting the screen that shows up when print is done, a list of installed filaments, my setting of PJH.

[PJH_issue_1.pdf](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7582836/PJH_issue_1.pdf)
[PJH_issue_2.pdf](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7582839/PJH_issue_2.pdf)
[PJH_issue_3.pdf](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7582841/PJH_issue_3.pdf)

OllisGit, 2022-03-18:
Hi @schumsel293,
I don't have a MMU and it looks like that other people have similar issues when using PJH and/or SpoolManager  (e.g. https://github.com/OllisGit/OctoPrint-SpoolManager/issues/251)

In the latest release V1.7.0  is a technical LogViewer (https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.17.0.dev1), Maybe that helps to analyse the issue.

bdaroz, 2022-03-27:
Here's the log file from a print I just ran.

Tool 0 had some Black filament, Tool 4 the white filament that was used. Both filaments were listed as active in their appropriate places in the SpoolManager. On start of print tool 4 was selected (Filament 5 on the MMU) as this print job is for a single filament.

SpoolManager selections. (Taken after the print, but unchanged)
[image]

Note the spool listed in the General tab is wrong:
[image]

Interesting it found the selected filement in tool0, but nothing in tool4 -- but it did see tool 4 used.
[image]

Technical log:

```
2022-03-27 11:21:52,231 - INFO - PrintJob '1ea_Glasses_0.2mm_PLA_MK3SMMU2S_1h32m.gcode' started!
2022-03-27 11:21:52,351 - INFO - Try reading Temperature from PreHeat-Plugin...
2022-03-27 11:21:52,363 - INFO - ... Temperature found '60.0' Tool 'tool0' '215.0'
2022-03-27 12:55:53,183 - INFO - PrintJob finished!
2022-03-27 12:55:53,184 - INFO - Print result:success, CaptureMode:always
2022-03-27 12:55:53,191 - INFO - ----- Start capturing print job data... -----
2022-03-27 12:55:53,198 - INFO - Start parsing Slicer-Settings
2022-03-27 12:55:53,519 - INFO - Finished parsing Slicer-Settings
2022-03-27 12:55:53,520 - INFO - ----- Start grab Image/thumbnail... -----
2022-03-27 12:55:53,521 - INFO - Try reading Thumbnail
2022-03-27 12:55:53,524 - WARNING - Thumbnail not found in print metadata
2022-03-27 12:55:53,525 - WARNING - Thumbnail not found for cameraManager
2022-03-27 12:55:53,525 - INFO - Try capturing snapshot asyc
2022-03-27 12:55:53,526 - INFO - Try taking snapshot '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20220327-112152.jpg' from 'http://127.0.0.1:8080/?action=snapshot'
2022-03-27 12:55:53,526 - INFO - ----- Start reading filament -----
2022-03-27 12:55:53,531 - INFO - Try reading filament from SpoolManager...
2022-03-27 12:55:53,567 - INFO - Process snapshot image
2022-03-27 12:55:53,609 - INFO -  reading for 'tool0',  Spool: 'Eryone Black Sparkle PLA', Material: 'PLA', Vendor: 'Eryone'
2022-03-27 12:55:53,610 - INFO -  reading for 'tool4',  Spool: 'Hatchbox White PLA', Material: 'PLA', Vendor: 'Hatchbox'
2022-03-27 12:55:53,802 - INFO - tool0: usedLength='0.0'; usedWeight='0.0'; usedCost='0.0'
2022-03-27 12:55:53,802 - INFO - tool1: usedLength='0.0'; usedWeight='0.0'; usedCost='None'
2022-03-27 12:55:53,804 - INFO - tool2: usedLength='0.0'; usedWeight='0.0'; usedCost='None'
2022-03-27 12:55:53,806 - INFO - tool3: usedLength='0.0'; usedWeight='0.0'; usedCost='None'
2022-03-27 12:55:53,807 - INFO - tool4: usedLength='4819.0981599997785'; usedWeight='0.0'; usedCost='None'
2022-03-27 12:55:53,808 - INFO - total: usedTotalLength='4819.0981599997785'; usedTotaWeight='0.0'; usedTotalCost='0.0'
2022-03-27 12:55:53,812 - INFO - ----- Start reading costs -----
2022-03-27 12:55:53,813 - INFO - Try calculating Costs with CostEstimation-Plugin settings
2022-03-27 12:55:53,814 - INFO - costPerHour '0.03' = powerConsumption '0.2' * costOfElectricity '0.15'
2022-03-27 12:55:53,815 - INFO - electricityCost '0.047007012975' = costPerHour '0.03' * printTimeInHours '1.5669004325'
2022-03-27 12:55:53,816 - INFO - printerCost '0.23895231595625' = (depreciationPerHour '0.0525' + maintenancePerHour '0.1') * printTimeInHours '1.5669004325'
2022-03-27 12:55:53,816 - INFO - Trying to read default filament values, for 'fallback-calculation'
2022-03-27 12:55:53,817 - INFO - No filament calculation for 'tool0', because usedLength is 0
2022-03-27 12:55:53,817 - INFO - No filament calculation for 'tool1', because usedLength is 0
2022-03-27 12:55:53,817 - INFO - No filament calculation for 'tool2', because usedLength is 0
2022-03-27 12:55:53,817 - INFO - No filament calculation for 'tool3', because usedLength is 0
2022-03-27 12:55:53,818 - INFO - No filament diameter from filamenttracker, using default '1.75'
2022-03-27 12:55:53,824 - INFO - No filament densityOfFilament from filamenttracker, using default '1.32'
2022-03-27 12:55:53,826 - INFO - No filament costOfFilament from filamenttracker, using default '25.0'
2022-03-27 12:55:53,826 - INFO - No filament weightOfFilament from filamenttracker, using default '1000.0'
2022-03-27 12:55:53,827 - INFO - Filament cost calculation for 'tool4'
2022-03-27 12:55:53,827 - INFO - Cost per weight '0.025' =  costOfFilament '25.0' / weightOfFilament '1000.0'
2022-03-27 12:55:53,829 - INFO - filamentCost '0.3825125521814367' = costPerWeight '0.025' * volumeWeight '15.300502087257467'
2022-03-27 12:55:53,829 - INFO - totalcost = '0.6684718811126866'
2022-03-27 12:55:53,830 - INFO - Adding costs from CostEstimation-Plugin: {'totalCosts': 0.6684718811126866, 'filamentCost': 0.3825125521814367, 'electricityCost': 0.047007012975, 'printerCost': 0.23895231595625, 'withDefaultSpoolValues': True}
2022-03-27 12:55:53,834 - INFO - ----- Try storing printjob model ----
2022-03-27 12:55:55,254 - INFO - ----- ... End PrintJob captured! -----
```

bdaroz, 2022-04-27:
Is there any more info I can get to help resolve this?

bdaroz, 2022-05-29:
Checking in....

bdaroz, 2022-06-29:
Updating to prevent closure.

bdaroz, 2022-08-02:
Still willing to provide additional information if needed. (Preventing closure)