OllisGit/OctoPrint-PrintJobHistory #182: Can't turn on Filament tracker

Mathew-D, 2021-11-27:
The cost will not auto add to the history anymore.  When I try and turn on the tracking I get a mouse curror with an X. I have both Filament tracking ad's installed and the cost shows up on other sections of Octoprint.   Any idea why it will not allow my to turn on either plugin for history?

I am running the newest version on the addons.

[image]

[image]

[image]
[image]
[image]

OllisGit, 2022-01-16:
Hi @Mathew-D,
please update to SpoolManager 1.6.0 and try again.

Maybe there is a JavaScript-Bug. Please open the Browser-Dev Tools Console (press F12) there should be no red messages.

closed 2022-02-03 by Mathew-D

lightmaster, 2022-02-13:
I'm seeing this issue with PrintJobHistory v1.16.0 and SpoolManager v1.6.1. Spool Manager is greyed out. The button had already been selected in a previous version, so it's still selected and shows the filament data in Print Job History. Will check after this print to make sure that Spool Manager is adjusting remaining filament correctly.