OllisGit/OctoPrint-PrintJobHistory #183: Cost Estimation aborted prints

EvilBlood, 2021-11-28:
Hi,

I use PrintJobHistory in combination with Spool Manager and Cost Estimation.

I have noticed that if a Print was aborted the Duration and used Filament ist updated but the Cost Stay the same as the print would finished.  

Versions:
Octoprint Version 1.7.2
Cost Estimation (3.4.0)
SpoolManager Plugin (1.5.0)
PrintJobHistory (1.15.2)

EvilBlood, 2021-11-29:
Hi,
I also found the after a successful print. The Costs would not update. (based on the real print time)

AlexVercammen, 2021-12-21:
In my case, sometimes the cost is not updated as well but randomly. I made sure CostEstimation was setup correctly. I am currently running a print and I will confirm my "random" statement...

OllisGit, 2022-01-16:
I think I found the issue. The Cost-Plugin doesn't update the data reliable.
I create a new ticket for that in the Cost-Repo: https://github.com/OllisGit/OctoPrint-CostEstimation/issues/16

OllisGit, 2022-02-14:
Hi @AlexVercammen, @EvilBlood,

I release a new version https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.17.0.dev1 in **DEV channel.**  

This version should fix the cost-bug and also add the option to see how the costs were calculated in a technical log.

 Please test and give feedback.

Thx, in advance
Olli

OllisGit, 2022-03-18:
fyi: Now included in V1.17.0

closed 2022-04-28 by github-actions[bot]