OllisGit/OctoPrint-PrintJobHistory #184: No Print Job History since tHE 26th

Filmman586, 2021-11-28:
All of a sudden I lost my print job history. I deleted the database and recreated, still have nothing. Attached my log File nothing else you need?
[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7614179/octoprint.log)

Filmman586, 2021-12-03:
I still can't get a history, I even reinstalled everything. Please help...

What do you need from me

Filmman586, 2021-12-09:
I think there is an issue with Pruser Slicer Thumbnails and Print Job History. I have both Cura Thumbnails and Prusa Slicer Thumbnails installed. I recently installed Prusa Slicer to try it out and thats when the problem started. I installed Prusa Slicer Thumbnails and I had no reports being generating, I disabled Prusa Slicer Thumbnails and my reports are back. Could there be a conflict with the 2?

OllisGit, 2022-01-16:
Hi @Filmman586,

please attach a prusa sliced gcode with the included thumbnail-image. It looks lie that the prusa-slicer changed the format and what version of prusasclicer do you use?

I tested with old prusa-gcode file and it works.

OllisGit, 2022-01-16:
With the following description and CURA I was able to reproduce the issue. Thx, @net147 for providing the link.
https://github.com/mriscoc/Marlin_Ender3v2/wiki/How-to-generate-a-gcode-preview

I will work on a solution...

OllisGit, 2022-01-22:
Hi @Filmman586,

new release is out and fixed the issue.

Please test and give feedback.

Thx, in advance
Olli

net147, 2022-01-23:
Works for me

Filmman586, 2022-01-23:
Olli,

Tested it. I've attached a screen shot. Both settings in Cura Thumbnails and Slicer Thumbnails are the same. And the background is black, and scaling is off. Take a look...

OllisGit, 2022-01-24:
@net147 sending emails with attachments to GitHub-Issues is not working. You need to attach the images directly in the GitHub-UI.

Filmman586, 2022-01-24:
Olli,

Tested it. I've attached a screen shot. Both settings in Cura Thumbnails and Slicer Thumbnails are the same. And the background is black, and scaling is off. Take a look...

[image]

closed 2022-02-14 by OllisGit