OllisGit/OctoPrint-PrintJobHistory #186: add Nozzel change button and field that summarizes all Printtimes till last nozzle change

Doprintityourself, 2022-01-08:
It would be very nice if we could track the total printtime of a nozzle to know how many hours the last change was.

So If I change to a new nozzle I could just hit a button to reset the total printtime for the new nozzle.
Therfore I can track how much time has been printed and in advance change the nozzle after the time period
I have set to have always the same print quality

Advanced would be to track different nozzles if to change between sizes by example numbers !?

Thank you

OllisGit, 2022-01-16:
Hi @Doprintityourself,

this is a good idea and I am planing a solution for this....but it take some time.

OllisGit, 2022-03-18:
Related to: https://github.com/OllisGit/OctoPrint-MaintenanceManager