OllisGit/OctoPrint-PrintJobHistory #187: Filament tracking not working

bambuleee, 2022-01-12:
Spool is selected for printing.
[image]

But in the finished print dialog it's not recognized. Stop working from one day to another.
Reinstalled PrintJobHistory plugin and Sppol Manager.

[image]

Tried with different spools.

OllisGit, 2022-01-16:
Hi,
I found in the logs that the needed filament was not provided by octoprint. That is the reason why the spool information is not tracked.
```
INFO - There is no calculated filament data in meta-file
```
The calculation is done during an "analyse-phase". The result is stored in the upload folder like this:
```
<OP>/uploads/metadata.json
  "3DBenchy.gcode": {
    "DisplayLayerProgress": {
      "totalLayerCountWithoutOffset": "159"
    },
    "analysis": {
      "estimatedPrintTime": 3529.315478428196,
      "printingArea": {
        "maxZ": 48.05,
        "maxX": 119.791,
        "maxY": 122.755,
        "minX": 20.0,
        "minY": 50.0,
        "minZ": 0.3
      },
      "dimensions": {
        "width": 99.791,
        "depth": 72.755,
        "height": 47.75
      },
      "filament": {
        "tool0": {
          "volume": 10.928834037976793,
          "length": 4543.68119999991
        }
      }
    },
```
Please check if the "analyse-feature" is activated like this:
[image]

bambuleee, 2022-01-17:
Hi,
the setting was not checked on always, but this with "always, it is not working
[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7881441/octoprint.log)
.

bambuleee, 2022-01-18:
I did a new installation (switch from octopi to octoprint-docker) with new settings, only backup from spool manager.
It is working now.

closed 2022-01-22 by OllisGit