OllisGit/OctoPrint-PrintJobHistory #193: Feature Request: Option to record print as a failure

jDavidnet, 2022-02-10:
Two fail modes I've experienced when printing are:
1. the filament wasn't loaded correctly or the nozzle get's jammed and the print never completed by morning.
2. there was a bridging or support failure and the print is just a bunch of filament spaghettis.

(3.) There is kind of a 3rd print failure, where the print succeeds but just looks bad.

I'd like to be able to mark in the print history what happened to that print.

OllisGit, 2022-02-10:
Currently there are only two status which could be edit after a print. Just click on the status-icon top left, select your status and click on save.

To cover more things like this request, I would like to implement a labels field. 
With the labels field you can assign your own custom values e.g. ```result:stringing```,```result:perfect``` or ```sold:yes```....then you are free to mark your jobs as you like.
[image]

This is planed...and not scheduled yet!

jDavidnet, 2022-02-10:
I'm willing to try it as a solution.  However, Failure is a very specific type of taxonomy that might affect the UI in the future.  For example a failed print might have it's text in red in a list.

jDavidnet, 2022-03-13:
Feature Requests and issues shouldn't automatically close

closed 2022-04-24 by github-actions[bot]