OllisGit/OctoPrint-PrintJobHistory #200: Plugin triggers a "please reload" dialog after every job

dimo414, 2022-08-02:
I've been noticing for a while now that if I leave OctoPrint open for too long it ends up displaying a "Please Reload" overlay dialog even though nothing should have changed with the server. A few searches found similar reports like https://github.com/OctoPrint/OctoPrint/issues/2950 which suggest the issue is plugins writing to the OctoPrint config unnecessarily. After snapshotting my config for a little while I hit another dialog today and this plugin appears to be a culprit:

```
$ diff -u /tmp/octoprint-config-2022-07-31_05.53.57.yaml /tmp/octoprint-config-2022-08-02_05.59.26.yaml
--- /tmp/octoprint-config-2022-07-31_05.53.57.yaml      2022-07-31 05:53:57.356692508 +0000
+++ /tmp/octoprint-config-2022-08-02_05.59.26.yaml      2022-08-02 05:59:26.685099734 +0000
@@ -34,7 +34,7 @@
     pluginCheckActivated: false
     preferedImageSource: camera
     selectedFilamentTrackerPlugin: SpoolManager Plugin
-    showPrintJobDialogAfterPrint_jobId: 251
+    showPrintJobDialogAfterPrint_jobId: 253
   SlicerEstimator:
     _config_version: 2
     pdp: '1'
```

I'm guessing this job ID value should be stored somewhere other than the OctoPrint config, since updating it seems to trigger this "please reload" overlay unnecessarily.

dimo414, 2022-09-02:
AFAIK this is still an issue.

dimo414, 2022-11-04:
@OllisGit I don't mind this bug sitting idle, I'm sure you have better things to do, but could you turn off the auto-closer? I don't know what else to do than ping this every month which is just obnoxious for everyone involved.

david-dasilva, 2023-04-05:
Please store the current job id elsewhere to prevent this reloading. 
I'm not familiar with Octoprint plugins but I will try to take a look when I find the time.