OllisGit/OctoPrint-PrintJobHistory #207: Images no longer appear after job completion

ailireruf, 2022-11-20:
I am using Slicer Thumbnails along with PrintJobHistory. It very convenient to visualize parts that are or were printer. 
In Prucaslicer The G-code thumbnail is set to PNG, 16x16 and 200x200. Previously there were no issues whatsoever - you could see the thumbnail in "files" section as well as an image in print job tab when the job was finishes. However now the thumbnails  only can be seen in 'files' section, but when the job is finished the image does not come up. 
No setting have been changed, neither on Octopi or in Prusaslicer 2.5.0. The images just do not appear anymore and I don't know why.
Is it a bug? It was working flawlessly previously and for some reason it stopped recognized the thumbnail as image or maybe its something else?
Thank you!

Tinkerman61, 2022-11-27:
I am having the same issue.  Since upgrading to Prusaslicer 2.5.0, thumbnail images no longer appear in the Print Job History log.  The Slicer-Settings for the print job still show the thumbnail lines.

[image]

JLGBJ, 2023-01-06:
I`m having the same problem
[plugin_PrintJobHistory_singlePrintJob.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/10359780/plugin_PrintJobHistory_singlePrintJob.log)

halofx, 2023-01-17:
I was having the same problem, and then noticed that occasionally the images would show up. After some testing, I found that the stock MK3 profiles worked, but Mini, and my existing ones weren't. The difference was that the stock MK3 profiles have gone to using just a single image size, I had 2 sizes in my profiles. Once I changed the dimensions to a single thumbnail size, they began working again. I am currently using 220x124.

ailireruf, 2023-01-20:
I tried doing that, and it didn't work. Could you please share more details? What g-code flavor are you using?
It's been like 2 month now, unfortunately still no solution :(

halofx, 2023-01-22:
I rolled everything back to the stock profile, and that is how I discovered the issue. That one change to a single size is all it was. 
The Mini is using Marlin 2, and the MK3 uses Legacy. They all use PNG, seems I remember JPG doesn't work? Don't remember.

It also seems I remember changing Analyze Gcode to Always under the Features in Octoprint.

Doprintityourself, 2023-02-10:
Same problem here, switching to single does not help analyze gcode is on always.
Compared to a old file before 2.5 it hast the same two png pictures and sizes in the gcode and it is displayed within Print Job History, but whatever I do it does not show up there now.
Any other last findings!?

dojohnso, 2023-10-19:
filenames with spaces and other non-standard filename characters break thumbnails. check my latest PR 

https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226

The author of this repo doesn't seem to pay attention anymore but you can update it manually per the steps I listed there.