OllisGit/OctoPrint-PrintJobHistory #214: Thumbnail images don't work in OctoPrint 1.9 release candidates

gbeatty, 2023-05-10:
The OctoPi builds of OctoPrint 1.9 release candidates 1 through 5 all seem to cause problems for the thumbnail images in PrintJobHistory. All thumbnails show up as the broken link image. Rolling back to 1.8 stable fixes the problem

pandel, 2023-05-17:
+1

Flask has deprecated the `cache_timeout` parameter in `send_file` method, which still seems to be used in the current PrintJobHistoryAPI (line 443).

Hopefully this will be removed soon.

dojohnso, 2023-05-25:
I can confirm @pandel 's solution above resolves the issue of thumbnails not showing up in Octoprint 1.9 (official release version for me)

dojohnso, 2023-05-25:
Not sure if this is still being maintained, but I created a PR to fix this

PR: https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/216

clarkiej, 2023-05-28:
I had teh same problem, and I made the change to PrintJobHistoryAPI.py, but my Pi is running Python 3.9: 
[image]

The change brought back all of the broken thumb nails:
[image]

but any new print just shows "No image available" even though the preview is in the gcode and can be read by the file listing.

pandel, 2023-05-29:
@clarkiej strange, that must be something different. In my case, the fix did not interfere with new thumbnails, they appear just fine.

clarkiej, 2023-06-02:
After 5 days of "No Image Available" the last two prints I have done have added the image to PrintJobHistory. No rhyme or reason, nothing has been changed, no updates run. Fingers crossed it has sorted itself out internally. I will let you know if it continues.
[image]

eacxx, 2023-06-03:
One more happy user :)
OctoPrint release 1.9.0
Thank you @pandel

JeremiahGillis, 2023-06-29:
> but any new completed print just shows "No image available" even though the preview is in the gcode and can be read by the file listing:

I have this exact same issue with OctoPrint 1.9.1 using the new camera stack. My camera stream and snapshot test work fine in the `Classic Webcam` plugins in settings.

I applied the fix to make my old thumbnails display. All new jobs do not display the thumbnails. They do no not exist in the snapshots directory.

[image]

If I click on the newest entry on the picture above, capture an image, press Save, and the picture still does not display, or appears in the snapshots directory.

[image]

After pressing Save:

[image]

Here is a log:

```
2023-06-28 18:05:53,746 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-06-28 18:06:50,282 - octoprint.printer.standard.job - INFO - Print job deselected - user: n/a
2023-06-28 18:06:55,770 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode
2023-06-28 18:06:55,797 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode
2023-06-28 18:06:55,827 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode, owner: _api, user: _api
2023-06-28 18:06:55,915 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2023-06-28 18:06:55,958 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode, owner: _api, user: _api
2023-06-28 18:06:56,215 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2023-06-28 18:06:56,264 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2023-06-28 18:06:56,323 - octoprint.plugins.PrintJobHistory - INFO - PrintJob 'Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode' started!
2023-06-28 18:06:56,366 - octoprint.plugins.PrintJobHistory - INFO - Try reading Temperature from PreHeat-Plugin...
2023-06-28 18:06:56,388 - octoprint.plugins.PrintJobHistory - WARNING - ... PreHeat Button Plugin not installed/enabled
2023-06-28 18:06:56,444 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2023-06-28 18:06:59,292 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2023-06-28 18:07:56,405 - octoprint.plugins.PrintJobHistory - INFO - Temperature read from Printer Bed: '60.0' Tool tool0: '230.0' after a delay of '60' seconds
2023-06-28 18:20:53,748 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-06-28 18:35:53,749 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-06-28 18:49:50,152 - octoprint.util.comm - INFO - Finished in 2574.264 s.
2023-06-28 18:49:50,153 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2023-06-28 18:49:50,162 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode
2023-06-28 18:49:50,184 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode, owner: _api
2023-06-28 18:49:50,209 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode
2023-06-28 18:49:50,359 - octoprint.plugins.PrintJobHistory - INFO - PrintJob finished!
2023-06-28 18:49:50,377 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2023-06-28 18:49:50,385 - octoprint.plugins.PrintJobHistory - INFO - ----- Start capturing print job data... -----
2023-06-28 18:49:50,417 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2023-06-28 18:49:50,472 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2023-06-28 18:50:00,483 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Mexican Train Dominos - Train Figure_0.6n_0.15mm_PLA_MK3S_42m.gcode finished, needed 10.32s
2023-06-28 18:50:00,746 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2023-06-28 18:50:00,747 - octoprint.plugins.PrintJobHistory - INFO - ----- Start grab Image/thumbnail... -----
2023-06-28 18:50:00,751 - octoprint.settings - WARNING - DeprecationWarning: Detected access to deprecated settings path ['webcam', 'snapshot'], returned value is derived from compatibility overlay. Please use the webcam system introduced with 1.9.0, this compatibility layer will be removed in a future release.
2023-06-28 18:50:00,770 - octoprint.plugins.PrintJobHistory - INFO - Try reading Thumbnail
2023-06-28 18:50:00,774 - octoprint.plugins.PrintJobHistory - WARNING - Thumbnail not found in print metadata
[... cut ...]
2023-06-28 18:50:01,017 - octoprint.plugins.filamentmanager - INFO - Updated remaining filament on spool 'Green Grass - Pro PLA (3D-Fuel)' from 994.6319126343805g to 990.443816626269g (-4.188096008111415g)
2023-06-28 18:50:02,226 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-06-28 18:50:02,270 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-06-28 18:50:53,751 - octoprint.server.heartbeat - INFO - Server heartbeat <3
```

JeremiahGillis, 2023-06-29:
Very strange, now it is working correctly on the fourth print. I did not install any new plugins or reboot the system. Here are updated logs with it working:

```
2023-06-28 19:49:05,184 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State:
| PreHeat=missing (None)
| filamentmanager=enabled (1.9.1)
| DisplayLayerProgress=missing (None)
| SpoolManager=missing (None)
| UltimakerFormat=missing (None)
| PrusaSlicerThumbnail=missing (None)
| costestimation=missing (None)

2023-06-28 19:49:05,298 - octoprint.server.util.flask - INFO - Passively logging in user ggoldfingerd from 192.168.1.123
2023-06-28 19:49:05,299 - octoprint.access.users - INFO - Logged in user: ggoldfingerd
2023-06-28 19:49:09,936 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-06-28 19:49:10,123 - octoprint.server.util.sockjs - INFO - User ggoldfingerd logged in on the socket from client 192.168.1.123
2023-06-28 19:49:10,866 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-06-28 19:49:20,885 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: RoofWireClips.gcode, owner: ggoldfingerd, user: ggoldfingerd
2023-06-28 19:49:20,974 - octoprint.util.comm - INFO - Starting job on behalf of user ggoldfingerd
2023-06-28 19:49:20,983 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2023-06-28 19:49:20,995 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: RoofWireClips.gcode, owner: ggoldfingerd, user: ggoldfingerd
2023-06-28 19:49:21,103 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2023-06-28 19:49:21,229 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2023-06-28 19:49:21,283 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2023-06-28 19:49:21,280 - octoprint.plugins.PrintJobHistory - INFO - PrintJob 'RoofWireClips.gcode' started!
2023-06-28 19:49:24,003 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2023-06-28 19:49:24,075 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2023-06-28 19:49:38,564 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2023-06-28 19:50:21,389 - octoprint.plugins.PrintJobHistory - INFO - Temperature read from Printer Bed: '60.0' Tool tool0: '230.0' after a delay of '60' seconds
2023-06-28 19:56:44,411 - octoprint.util.comm - INFO - Finished in 443.436 s.
2023-06-28 19:56:44,413 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2023-06-28 19:56:44,430 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: RoofWireClips.gcode, owner: ggoldfingerd
2023-06-28 19:56:44,573 - octoprint.plugins.PrintJobHistory - INFO - PrintJob finished!
2023-06-28 19:56:44,583 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2023-06-28 19:56:44,595 - octoprint.plugins.PrintJobHistory - INFO - ----- Start capturing print job data... -----
2023-06-28 19:56:44,628 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2023-06-28 19:56:44,644 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2023-06-28 19:56:44,979 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - DEBUG -  Slicer-Settings:
2023-06-28 19:56:44,980 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - DEBUG - {'generated by': ' generated by PrusaSlicer 2.6.0+win64 on 2023-06-29 at 01:07:26 UTC\n', 'external perimeters extrusion width': '0.65mm', 'perimeters extrusion width': '0.65mm', 'infill extrusion width': '0.65mm', 'solid infill extrusion width': '0.65mm', 'top infill extrusion width': '0.60mm', 'first layer extrusion width': '0.65mm', 'prusaslicer_config': 'end', 'z_offset': '0', 'xy_size_compensation': '0', 'wiping_volumes_matrix': '0', 'wiping_volumes_extruders': '70,70', 'wipe_tower_y': '125', 'wipe_tower_x': '170', 'wipe_tower_width': '60', 'wipe_tower_rotation_angle': '0', 'wipe_tower_no_sparse_layers': '0', 'wipe_tower_extruder': '0', 'wipe_tower_extra_spacing': '100%', 'wipe_tower_cone_angle': '0', 'wipe_tower_brim_width': '2', 'wipe_tower_bridging': '10', 'wipe_tower': '1', 'wipe_into_objects': '0', 'wipe_into_infill': '0', 'wipe': '1', 'wall_transition_length': '100%', 'wall_transition_filter_deviation': '25%', 'wall_transition_angle': '10', 'wall_distribution_count': '1', 'variable_layer_height': '1', 'use_volumetric_e': '0', 'use_relative_e_distances': '1', 'use_firmware_retraction': '0', 'travel_speed_z': '12', 'travel_speed': '180', 'travel_acceleration': '0', 'top_solid_min_thickness': '0.9', 'top_solid_layers': '6', 'top_solid_infill_speed': '45', 'top_solid_infill_acceleration': '0', 'top_infill_extrusion_width': '0.6', 'top_fill_pattern': 'monotoniclines', 'toolchange_gcode': '', 'thumbnails_format': 'PNG', 'thumbnails': '', 'threads': '32', 'thin_walls': '0', 'thick_bridges': '0', 'template_custom_gcode': '', 'temperature': '225', 'support_tree_top_rate': '15%', 'support_tree_tip_diameter': '0.8', 'support_tree_branch [... cut ...]
2023-06-28 19:56:45,334 - octoprint.plugins.filamentmanager - INFO - Updated remaining filament on spool 'Green Grass - Pro PLA (3D-Fuel)' from 989.7558552719253g to 989.0678939175816g (-0.6879613543436562g)
2023-06-28 19:56:45,518 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-06-28 19:56:45,653 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
```

Titius-Sicanus, 2023-07-01:
I've been having problems detecting thumbnails when printing was completed for a long time.
Some were mined while others were not.
Eventually I found that the thumbnail not extracting depends on the file name.
If the file name contains spaces or special characters then the thumbnail is not extracted.
The only characters I've found allowed are the minus sign (-) and the underscore character (_)

While I find the decision not to allow the presence of special characters in the file name to be partially acceptable, I find the ban on the use of space unacceptable.

EricZwirnmann, 2023-07-03:
> +1
> 
> Flask has deprecated the `cache_timeout` parameter in `send_file` method, which still seems to be used in the current PrintJobHistoryAPI (line 443).
> 
> See also: https://flask.palletsprojects.com/en/2.3.x/api/?highlight=send_file#flask.send_file
> 
> Hopefully this will be removed soon.
> 
> EDIT: In the meantime you can fix it yourself by editing the file (make a backup before)
> 
> `/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py`
> 
> Change line 443 to this:
> 
> `return send_file(absoluteFilename, mimetype='image/jpg')`
> 
> Be careful to not remove any whitespace before or after the line! Just remove the `, cache_timeout=1` part and restart Octoprint. You're done!

After putting in this fix, I found that my history images are now upside down (flipped).  I do have the setting for my camera set to flip horizontally and vertically so that they appear correctly when I monitor my camera.  The new images stored and my history images are now upside down.

[image]

morris2009, 2023-07-31:
FYI: The fix worked for me!
tnx!

HarikrishnanBalagopal, 2023-08-02:
> +1
>
> Flask has deprecated the `cache_timeout` parameter in `send_file` method, which still seems to be used in the current PrintJobHistoryAPI (line 443).
>
> See also: [...]

I have the same issue, here's the logs from `/home/pi/.octoprint/logs/octoprint.log`

```
2023-08-02 11:55:48,268 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/printJobSnapshot/20230802-113600.jpg [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 443, in get_snapshot
    return send_file(absoluteFilename, mimetype='image/jpg', cache_timeout=1)
```

HarikrishnanBalagopal, 2023-08-02:
> 

Depending on your Python version you will have to change the path
`/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py`
to
`/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py`
for Python 3.9 for example

dojohnso, 2023-08-02:
I'm told @OllisGit is back from a dev sabbatical so hopefully my PR will see some movement, along with other bugs.

dojohnso, 2023-10-19:
@gbeatty @pandel @clarkiej 

Sorry for the spam hit but wanted to make sure folks see this. I think i solved the new issue found around file names breaking thumbnails. please see this PR and comment as necessary. https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226

can't wait to see what's next that I need to fix for this..