OllisGit/OctoPrint-PrintJobHistory #218: After update to Octoprint V1.9.0 all pictures are not shown

Hottabych-, 2023-06-04:
When updating to Octoprint the captured pictures of the prints aren't shown any more:
In the detail view occurs the same.
[image]

impossibleturnip, 2023-06-06:
This is also happening for me, I'm not seeing any the images, just the broken link image. This happens for all of my print job history, including those performed prior to the upgrade to 1.9.0. If i use the browser inpspector to look at the HTML code and copy the URL of one of the broken images into a new tab, to see if the image can be accessed, I get the following error:
{
  "error": "The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."
}

Hottabych-, 2023-06-06:
I've added two logs.
--> octoprint (PrintJobHistory).log this is an excerpt from this log: octoprint.log

[octoprint (PrintJobHistory).log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/11668529/octoprint.PrintJobHistory.log)
[plugin_PrintJobHistory_singlePrintJob.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/11668530/plugin_PrintJobHistory_singlePrintJob.log)

Ricky0351, 2023-06-09:
same issue for me. to follow.

Hottabych-, 2023-06-09:
Hi all,
I'll got a kind hint from the user _Trashware_ from the  OctoPrint Community Forum. He send the reference to this temporary solution: https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/214 . I can say this hint work and solve the issue until we get an official fix. 
You have to adapt the hint to the new directory of python:
/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py

Best 
'the wizard' ;-)

Erny52, 2023-06-10:
Many thanks to all for sharing the fix for solving the 'image issue' in PrintJobHistory. Works perfectly.
Best regards,
Erny52

jDavidnet, 2023-06-27:
Octoprint crashed for me, and it seems to be picture/snapshot related.

```
2023-06-26 23:34:53,198 - octoprint.access.users - INFO - Cleaning up user session 453763B1AE454C658DC63B4389558931 for user admin
2023-06-26 23:34:53,198 - octoprint.access.users - INFO - Logged out user: admin
2023-06-26 23:34:53,208 - octoprint.server.util.flask - INFO - Passively logging in user admin from ::ffff:192.168.243.236
2023-06-26 23:34:53,210 - octoprint.access.users - INFO - Logged in user: admin
2023-06-26 23:34:53,232 - octoprint.server.util.sockjs - INFO - User admin logged in on the socket from client ::ffff:192.168.243.236
2023-06-26 23:34:54,246 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/printJobSnapshot/20230626-134053.jpg [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 443, in get_snapshot
    return send_file(absoluteFilename, mimetype='image/jpg', cache_timeout=1)
TypeError: send_file() got an unexpected keyword argument 'cache_timeout'
2023-06-26 23:34:55,361 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 36899, 'printer_state': 'OPERATIONAL'}
2023-06-26 23:34:56,143 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 59, 'spoolName': 'Marble White Matte', 'material': 'PLA', 'colorName': 'Marble White', 'remainingWeight': 1013.7555693118106}' to event-bus
2023-06-26 23:34:56,168 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 59, 'spoolName': 'Marble White Matte', 'material': 'PLA', 'colorName': 'Marble White', 'remainingWeight': 1013.7555693118106}' to event-bus
2023-06-26 23:34:56,176 - octoprint.plugins.SpoolManager - INFO - tool0, requiredWeight '9.73315455238686',  remainingWeight '1013.7555693118106'
2023-06-26 23:49:43,244 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-06-26 23:49:55,564 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 37799, 'printer_state': 'OPERATIONAL'}
2023-06-27 00:04:19,942 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.243.236
2023-06-27 00:04:43,245 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-06-27 00:04:55,455 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 38699, 'printer_state': 'OPERATIONAL'}
2023-06-27 00:08:13,500 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.243.236
2023-06-27 00:08:13,511 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State:
| PreHeat=enabled (0.8.0)
| filamentmanager=missing (None)
| DisplayLayerProgress=enabled (1.28.0)
| SpoolManager=enabled (1.7.0)
| UltimakerFormat=missing (None)
| PrusaSlicerThumbnail=enabled (1.0.2)
| costestimation=enabled (3.5.0)

2023-06-27 00:08:14,534 - octoprint.server.util.flask - INFO - Passively logging in user admin from ::ffff:192.168.243.236
2023-06-27 00:08:14,535 - octoprint.access.users - INFO - Logged in user: admin
2023-06-27 00:08:14,568 - octoprint.server.util.sockjs - INFO - User admin logged in on the socket from client ::ffff:192.168.243.236
2023-06-27 00:08:15,761 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/printJobSnapshot/20230626-134053.jpg [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 443, in get_snapshot
    return send_file(absoluteFilename, mimetype='image/jpg', cache_timeout=1)
TypeError: send_file() got an unexpected keyword argument 'cache_timeout'
2023-06-27 00:08:16,605 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 59, 'spoolName': 'Marble White Matte', 'material': 'PLA', 'colorName': 'Marble White', 'remainingWeight': 1013.7555693118106}' to ev [... cut ...]
2023-06-27 00:08:16,616 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 59, 'spoolName': 'Marble White Matte', 'material': 'PLA', 'colorName': 'Marble White', 'remainingWeight': 1013.7555693118106}' to event-bus
2023-06-27 00:08:16,631 - octoprint.plugins.SpoolManager - INFO - tool0, requiredWeight '9.73315455238686',  remainingWeight '1013.7555693118106'
2023-06-27 00:08:16,639 - octoprint.plugins.uicustomizer - INFO - Checking for updated themes
```

dojohnso, 2023-10-19:
for anyone following along... https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226

Check the fix in this PR, should take care of the filename issue. The author of this repo doesn't seem to pay attention anymore but you can update it manually per the steps I listed there.

There is also a fix in that PR (dupe of my first one) for the cache_timeout issue