OllisGit/OctoPrint-PrintJobHistory #219: No thumbnail in history list if spaces/accents in filename

PyBonnetainNesterenko, 2023-06-10:
(still using OctoPrint 1.8)

Hello,
The plugin works great, but the thumbnails are not displayed if the printed filename contains (at least) spaces and accents. Which is annoying.

Fist with accents :
[image]

A proper display and print :
[image]

Renamed the previous file to add some spaces (and printed/canceled it, it's only to show the problem) :
[image]

dojohnso, 2023-10-19:
https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226

Check the fix in this PR, should take care of this. The author of this repo doesn't seem to pay attention anymore but you can update it manually per the steps I listed there.

PyBonnetainNesterenko, 2023-10-22:
As far as my limited tests show, [this suggestion](https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226) solves the problem. Thanks !