OllisGit/OctoPrint-PrintJobHistory #224: Prusa Slicer Thumbnail

TMTYD, 2023-08-12:
The thumbnails are not saved in the Print job History, although I have installed the Prusaslicer Thumbnail plugin and have also specified the values for Thumbnail in PrusaSlicer under Print settings. Why is that?

[image]

[Technical Log](https://privatebin.net/?cad487d3efe44c54#2dJMmtRURwm2Kv9cAKMCnxnfaZq1uD4jrRQWiNSvXceP)

TMTYD, 2023-10-03:
Hi, I changed that, but unfortunately no improvement. What size must be set in the Prusa slicer for G-code thumbnails? I set 16x16, 200x200 there and png as the format

dojohnso, 2023-10-05:
> [plugin_PrintJobHistory_singlePrintJob.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/12790139/plugin_PrintJobHistory_singlePrintJob.log) Hi, I changed that, but unfortunately no improvement. What size must be set in the Prusa slicer for G-code thumbnails? I set 16x16, 200x200 there and png as the format

that's what mine is set to as well. i've noticed this fix doesn't work for everyone, even on one of my Pi's it's not working but I haven't gotten into why yet. Likely will have to do with versions of board or OS or python, etc.

TMTYD, 2023-10-16:
Hi why does it sometimes work and sometimes not?

[image]

Titius-Sicanus, 2023-10-16:
If the file name contains spaces or special characters then the thumbnail is not extracted.
The only characters I've found allowed are the minus sign (-) and the underscore character (_)

TMTYD, 2023-10-16:
Ok, can you change that somehow?

Titius-Sicanus, 2023-10-16:
I'm sorry but I'm also waiting for someone to be able to solve this problem.

dojohnso, 2023-10-17:
If octoprint ever acknowledges this plugin is abandoned and someone takes it over (I've been trying for months), best I can do is take a look and maybe submit another PR that people can see and copy the changes locally.

TMTYD, 2023-10-19:
> If the file name contains spaces or special characters then the thumbnail is not extracted.
> The only characters I've found allowed are the minus sign (-) and the underscore character (_)

So that works with the _ substir, then the thumbnail is also saved

dojohnso, 2023-10-19:
@TMTYD @Titius-Sicanus 

https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226

closed 2025-04-08 by TMTYD