OllisGit/OctoPrint-PrintJobHistory #227: **Thumbnails broken? READ THIS first**

dojohnso, 2023-10-19:
https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226 https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/214

Check out the fixes in the PR(s) (226 includes 214). The author of this repo doesn't seem to pay attention anymore but you can update the files manually per the steps I listed on those. 

If they don't solve your issues, feel free to post a comment here and include the Technical Log from the print job history job having issues and I'll do my best to help.

RalfZi, 2023-11-01:
No thumbnail by saving the print history but in the log is no error

`2023-11-01 20:24:32,886 - INFO - ----- Start grab Image/thumbnail... -----
2023-11-01 20:24:32,905 - INFO - Try reading Thumbnail
2023-11-01 20:24:32,908 - INFO - Thumbnail is present
2023-11-01 20:24:32,909 - INFO - Thumbnail was captured from metadata
2023-11-01 20:24:32,910 - INFO - Try to take thumbnail, because afterprint/gcode not selected
2023-11-01 20:24:32,912 - INFO - Try reading Thumbnail
2023-11-01 20:24:32,916 - INFO - Try converting thumbnail '/home/pi/.octoprint/data/PrintJobHistory/../prusaslicerthumbnails/RODSLOTH_0.6n_0.2mm_PLA_MK3S_43m.png' to '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20231101-193037.jpg'
2023-11-01 20:24:32,924 - INFO - Converting successfull!
2023-11-01 20:24:32,925 - INFO - Thumbnail was captured from metadata
2023-11-01 20:24:32,927 - INFO - ----- Start reading filament -----
2023-11-01 20:24.

RalfZi, 2023-11-02:
I fix the error in correct line 443 in PrintJobHistoryAPI.py to 
return send_file(absoluteFilename, mimetype='image/jpg' )
greetings

ezi00, 2023-11-04:
> I fix the error in correct line 443 in PrintJobHistoryAPI.py to return send_file(absoluteFilename, mimetype='image/jpg' ) greetings

thanks worked for me as well..

Filmman586, 2023-11-18:
Will an update be coming anytime soon?

dojohnso, 2023-11-21:
anyone who sees this and the PRs mentioned above haven't been merged yet, please head over to this issue and comment on it to try and convince OctoPrint that this plugin has been abandoned. Cheers!

https://github.com/OctoPrint/plugins.octoprint.org/issues/1222

/cc @Filmman586

dojohnso, 2023-11-21:
> I fix the error in correct line 443 in PrintJobHistoryAPI.py to return send_file(absoluteFilename, mimetype='image/jpg' ) greetings

I'm glad it worked for you! Did you happen to see the change in this PR that changes the cache_timeout parameter to max_age? Just curious if that would've worked as well. Sounds like the change you made is an alternate approach to the same bug.

Filmman586, 2023-11-26:
I have no way of editing the file, can I get the file already fixed?

Hottabych-, 2023-11-26:
> I have no way of editing the file, can I get the file already fixed?
Don't understand, because if you can exchange the file then you can edit your local file too e.g. with nano oder vi. As you can read in several sources it is a very simple fix.  Look at https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/218#issuecomment-1584347340 alternativ https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226

Filmman586, 2023-11-26:
Thank you! I did I’ll give it a try!

Don't understand, because if you can exchange the file then you can edit your local file too e.g. with nano oder vi. As you can read in several sources it is a very simple fix. Look at #218 (comment)<https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/218#issuecomment-1584347340> alternativ #226<https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/226>

dojohnso, 2023-11-29:
WE DID IT - i now control this plugin and have pushed the update. thanks everyone!