OllisGit/OctoPrint-PrintJobHistory #231: CSV Export not working

spsijsr-rnd-kunald, 2024-05-23:
Upon clicking the CSV Export button the webpage shows that the link is loading and perpetually keeps buffering.

rtrout, 2024-06-27:
I'm also experiencing this problem. Have only recently added this plugin, so my first time attempting export.

I note the plugin calls URL: /plugin/PrintJobHistory/exportPrintJobHistory/CSV

It appears to just timeout.

Anywhere I can look to post logs?

RalfZi, 2024-11-09:
Hi I get the following error if I try Export to CSV

`sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was creat>
Nov 09 14:34:12 octoprint kernel: workqueue: drm_fb_helper_damage_work hogged CPU for >13333us 256 times, consider switching to WQ_UNBOUND
Nov 09 14:35:01 octoprint CRON[8463]: pam_unix(cron:session): session opened for user root(uid=0) by root(uid=0)
Nov 09 14:35:01 octoprint CRON[8464]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
Nov 09 14:35:01 octoprint CRON[8463]: pam_unix(cron:session): session closed for user root
Nov 09 14:35:18 octoprint octoprint[8246]: 2024-11-09 14:35:18,893 - octoprint.server.util.tornado - ERROR - Exception in WSGI application
Nov 09 14:35:18 octoprint octoprint[8246]: Traceback (most recent call last):
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/octoprint/server/util/tornado.py", line 782, in handle_request
Nov 09 14:35:18 octoprint octoprint[8246]:     chunk = await loop.run_in_executor(self.executor, next_chunk)
Nov 09 14:35:18 octoprint octoprint[8246]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/usr/lib/python3.12/concurrent/futures/thread.py", line 58, in run
Nov 09 14:35:18 octoprint octoprint[8246]:     result = self.fn(*self.args, **self.kwargs)
Nov 09 14:35:18 octoprint octoprint[8246]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/octoprint/server/util/tornado.py", line 777, in next_chunk
Nov 09 14:35:18 octoprint octoprint[8246]:     return next(app_response_iter)
Nov 09 14:35:18 octoprint octoprint[8246]:            ^^^^^^^^^^^^^^^^^^^^^^^
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/werkzeug/wsgi.py", line 500, in __next__
Nov 09 14:35:18 octoprint octoprint[8246]:     return self._next()
Nov 09 14:35:18 octoprint octoprint[8246]:            ^^^^^^^^^^^^
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/werkzeug/wrappers/response.py", line 50, in _iter_encoded
Nov 09 14:35:18 octoprint octoprint[8246]:     for item in iterable:
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/octoprint_PrintJobHistory/common/CSVExportImporter.py", line >
Nov 09 14:35:18 octoprint octoprint[8246]:     for job in allJobsDict:
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/peewee.py", line 4683, in next
Nov 09 14:35:18 octoprint octoprint[8246]:     self.cursor_wrapper.iterate()
Nov 09 14:35:18 octoprint octoprint[8246]:   File "/home/ralf/OctoPrint/lib/python3.12/site-packages/peewee.py", line 4594, in iterate
Nov 09 14:35:18 octoprint octoprint[8246]:     row = self.cursor.fetchone()
Nov 09 14:35:18 octoprint octoprint[8246]:           ^^^^^^^^^^^^^^^^^^^^^^
Nov 09 14:35:18 octoprint octoprint[8246]: sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was cre`

vojtakaniok, 2024-12-06:
Same here !

Log from octorpint:
```
2024-12-06 08:11:42,204 - octoprint.server.util.tornado - ERROR - Exception in WSGI application
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/tornado.py", line 782, in handle_request
    chunk = await loop.run_in_executor(self.executor, next_chunk)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/tornado.py", line 777, in next_chunk
    return next(app_response_iter)
  File "/home/pi/oprint/lib/python3.7/site-packages/werkzeug/wsgi.py", line 500, in __next__
    return self._next()
  File "/home/pi/oprint/lib/python3.7/site-packages/werkzeug/wrappers/response.py", line 50, in _iter_encoded
    for item in iterable:
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/common/CSVExportImporter.py", line 457, in transform2CSV
    for job in allJobsDict:
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 4408, in next
    self.cursor_wrapper.iterate()
  File "/home/pi/oprint/lib/python3.7/site-packages/peewee.py", line 4319, in iterate
    row = self.cursor.fetchone()
sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was created in thread id 1700729952 and this is thread id 1650373728.
```

rtrout, 2024-12-07:
Ok, my github detective fu is poor. This version of printjobhistory is abandoned, https://github.com/dojohnso/OctoPrint-PrintJobHistory is the latest fork but also abandoned, but probably where this issue should be reported (if it wasn't abandoned).

vojtakaniok, 2025-02-24:
@spsijsr-rnd-kunald Should be fixed in my new release, feel free to update